schoolpost/PiDNG #11: Not native resolution

chriskoups, 2020-06-16:
Really nice work!

I found that if I am not using the native resolution of my camera (ov_5647) the DNG becomes stripped with an alternating darker pixel. I prefer to use binned (2x2) images for better noise reduction.

I try to understand if the dng has always the native resolution and if it is possible to change that.

schoolpost, 2020-06-17:
You can upload your images then I might be able to see the exact issue. 

As far as I know, when you capture JPEG + RAW attached it is always using full sensor regardless of the sensor mode you have selected.

chriskoups, 2020-06-17:
If you extract the zip file you can find two images, one 2x2 binned and one full resolution with their DNG.
[Downloads.zip](https://github.com/schoolpost/PyDNG/files/4791682/Downloads.zip)

What you say it make sense, however when I open the DNG is the binned resolution so I am not sure what is extracting. Plus you will see that the image is dotted and not close to the original. I am using an OV_5647, I do not know if other cameras do the same.

chriskoups, 2020-06-18:
I might found a solution but I do not have time to explore it now.
If you have, can you test it with your camera?

As you correct said, RAW files are in full sensor always. However, when we make a binned file, the exif register the binned size and not the original. By rewriting the correct one, I was able to produce a good DNG.

I will suggest that you embed the native resolutions of those 3 cameras in the settings.
If you cannot do/test it, I will do it on Saturday. 
:)

closed 2020-06-19 by schoolpost