schoolpost/PiDNG #15: bit unpacking for the HQ camera is wrong

scriptorron, 2020-06-28:
Line 184 and following does the bit unpacking for the HQ camera:
```
            data = data.astype(np.uint16)
            data[:, 0::3] = data[:, 0::3] << 4
            data[:, 0::3] |= ((data[:, 1::3] >> 4) & 0x0f)
            data[:, 1::3] = data[:, 1::3] << 4
            data[:, 1::3] |= (data[:, 1::3] & 0x0f)
            data = np.delete(data, np.s_[2::3], 1)
```
This can't be: byte 2 is newer used and byte 1 is used too often. The documentation of the bit packing in [https://www.kernel.org/doc/html/v4.17/media/uapi/v4l/pixfmt-srggb12p.html](https://www.kernel.org/doc/html/v4.17/media/uapi/v4l/pixfmt-srggb12p.html) describes a format like this:
```
AAAAAAAA BBBBBBBB BBBBAAAA
```
The code for that is:
```
    data = data.astype(np.uint16)
    shape = data.shape
    unpacked_data = np.zeros((shape[0], shape[1] / 3 * 2), dtype=np.uint16)
    unpacked_data[:, ::2] = (data[:, ::3] << 4) + (data[:, 2::3] & 0x0F)
    unpacked_data[:, 1::2] = (data[:, 1::3] << 4) + ((data[:, 2::3] >> 4) & 0x0F)
```

scriptorron, 2020-06-29:
Sorry, in the my code above the `shape[1] /3 *2` leads to a `TypeError: 'float' object cannot be interpreted as an integer`. Correct is:
```
            data = data.astype(np.uint16)
            shape = data.shape
            unpacked_data = np.zeros((shape[0], int(shape[1] / 3 * 2)), dtype=np.uint16)
            unpacked_data[:, ::2] = (data[:, ::3] << 4) + (data[:, 2::3] & 0x0F)
            unpacked_data[:, 1::2] = (data[:, 1::3] << 4) + ((data[:, 2::3] >> 4) & 0x0F)
            data = unpacked_data
```

schoolpost, 2020-07-02:
Will implement in next release.

closed 2020-07-02 by schoolpost