schoolpost/PiDNG #16: I got this trying to run the example "picam.py"

inteblio, 2020-07-01:
This is run on a raspberry pi, python
Python 3.7.3
uname -a = 5.4.47-v7l

$ python3 picam.py
Traceback (most recent call last):
  File "picam.py", line 30, in <module>
    output = d.convert(stream)
  File "/home/pi/.local/lib/python3.7/site-packages/pydng/core.py", line 222, in convert
    rawFrame = self.__process__(image, process)
  File "/home/pi/.local/lib/python3.7/site-packages/pydng/core.py", line 195, in __process__
    rawImage = self.__extractRAW__(input_file)
  File "/home/pi/.local/lib/python3.7/site-packages/pydng/core.py", line 164, in __extractRAW__
    assert data[:4] == 'BRCM'.encode("ascii")
AssertionError

schoolpost, 2020-07-02:
Hello, can you tell me which camera you are using?

dwisted, 2020-07-02:
> Hello, can you tell me which camera you are using?

I also got the error I am using the High Quality Camera (imx477)

schoolpost, 2020-07-02:
> > Hello, can you tell me which camera you are using?
> 
> I also got the error I am using the High Quality Camera (imx477)

Can you try with python 3.6 and see if you get the same error?

schoolpost, 2020-07-02:
@dwisted @inteblio The example you are using assumes the OV5647 ( Camera Module V1 ) for its usage, not the newer High Quality Camera. This may be causing the issue.

inteblio, 2020-07-02:
Thank you!
Yes, I was using the High Quality Camera (imx477), Utility.py was not working for me either until I updated to pydng-3.1.2. 
Many thanks, I highly appreciate your response and work. 
Also, I used pip3 to install it on the raspberry pi, I think somebody on the forum got stumped by that. (used "pip") ?

closed 2020-07-02 by schoolpost