schoolpost/PiDNG #26: Unable to extract RAW from RPi-Cam-Web-Interface still image

dirkessl, 2020-09-22:
when trying to extract the raw file from a still image from RPi-Cam-Web-Interface, the following error appears:
pi@raspberrypi:~/PyDNG $ python3 examples/utility.py im.jpg
Traceback (most recent call last):
  File "examples/utility.py", line 54, in <module>
    main()
  File "examples/utility.py", line 51, in main
    d.convert(inputFilename, compress=compress)
  File "/home/pi/.local/lib/python3.7/site-packages/pydng/core.py", line 258, in convert
    rawFrame = self.__process__(image, process)
  File "/home/pi/.local/lib/python3.7/site-packages/pydng/core.py", line 223, in __process__
    rawImage = self.__extractRAW__(input_file)
  File "/home/pi/.local/lib/python3.7/site-packages/pydng/core.py", line 175, in __extractRAW__
    }[str(self.__exif__['Image Model'])]
KeyError: 'imx219'

when making a still with raspistill -r -o im.jpg it works. sample file is attached.

cheers,
dirk

[im.jpg.zip](https://github.com/schoolpost/PyDNG/files/5263844/im.jpg.zip)

schoolpost, 2020-09-25:
The camera is labelled differently in its metadata when using that application, you can simply tweak the source and add the specific label to the appropriate areas: 

Currently they are: "RP_imx219" you just need to add "imx219" instead.

https://github.com/schoolpost/PyDNG/blob/15264b14d5be6429d72d11b6aa12eb13b21cec26/src/pydng/core.py#L43-L59

dirkessl, 2020-09-25:
Most helpful, thank you!

closed 2020-09-25 by dirkessl