schoolpost/PiDNG #27: Lens Profiles

greg-randall, 2020-10-11:
I don't know if you'd like to include this kind of thing, but I'm working on making lens/camera profiles for the Raspberry Pi cameras that I have. Here's one for the HQ Camera Wide Angle Lens.
[hq_camera_wide_angle_profile.zip](https://github.com/schoolpost/PyDNG/files/5361795/hq_camera_wide_angle_profile.zip)
[image]

schoolpost, 2020-10-12:
I'll have to look into the DNG SDK and see how I can implement, but yea that would be very useful to have lens corrections.

greg-randall, 2020-10-12:
Sounds good. I'll generate profiles for the Raspberry Pi cameras/lenses I have. 

Is this the most convenient way for me to send you those files?

schoolpost, 2020-10-13:
Doesn't look like .lcpp files are something that can just be embedded into the DNG directly...

You can see how the expected lens corrections would be done using the OPCODE functions in the SDK. 
can be read about here: https://www.adobe.com/content/dam/acom/en/products/photoshop/pdfs/dng_spec_1.4.0.0.pdf

Chapter 7, specifically opcode 1 and opcode 2 would be for doing lens corrections.

closed 2020-11-11 by schoolpost