schoolpost/PiDNG #28: Convert from raspiraw

dustinkerstein, 2020-11-06:
PyDng is working great at converting my raspistill raw + jpg files. But now I'm looking more into https://github.com/6by9/raspiraw and was curious, is there any way to easily add support for the raw files (with header) that don't have the additional jpeg data? Thanks!

schoolpost, 2020-11-06:
This is currently possible as is, it would simply require a couple tweaks from the raw2dng example.

https://github.com/schoolpost/PyDNG/blob/15264b14d5be6429d72d11b6aa12eb13b21cec26/examples/raw2dng.py#L1-L49

dustinkerstein, 2020-11-07:
Nice! What kind of tweaks would be necessary to get it to match the color output from a normal jpg + raw conversion?

schoolpost, 2020-11-10:
> Nice! What kind of tweaks would be necessary to get it to match the color output from a normal jpg + raw conversion?

You would need to include these color matrices and the forward matrices and the calibration illumants in the DNG tags. This is how it's done internally in PyDNG with the HQ camera.

https://github.com/schoolpost/PyDNG/blob/15264b14d5be6429d72d11b6aa12eb13b21cec26/src/pydng/core.py#L286-L309

dustinkerstein, 2020-11-10:
Perfect, thank you!

stefandouw, 2020-12-11:
I have been trying to get it to work but I keep getting struct.errors:
struct.error: unpack requires a buffer of xxx bytes

What do I have to change in raw2dng.py to make it work with raspiraw files?

schoolpost, 2021-10-07:
Here is some sample code to convert your Bayer data to a DNG. Note: I'm not 100% sure this is being "unpacked" correctly, You will need to investigate this yourself. The color temperature and tint need to be adjusted manually, you need to provide more metadata about the Bayer raw to properly do white balancing from the get go. ( the R/B gains ) 

Code below is for the 3840, but likewise just change the width, height variables to the frame size for the 1080 versions. 

```
from pidng.core import RAW2DNG, DNGTags, Tag
import numpy as np
import struct, collections

width = 3840
height = 2160
bpp=10
numPixels = width*height
rawFile = 'test3840.raw'
rf = open(rawFile, mode='rb')
rawData = struct.unpack("H"*numPixels,rf.read(2*numPixels))
rawFlatImage = np.zeros(numPixels, dtype=np.uint16)
rawFlatImage[:] = rawData[:] 
rawImage = np.reshape(rawFlatImage,(height,width))

ccm1 = [[19549, 10000], [-7877, 10000], [-2582, 10000],	
        [-5724, 10000], [10121, 10000], [1917, 10000],
        [-1267, 10000], [ -110, 10000], [ 6621, 10000]]

t = DNGTags()
t.set(Tag.ImageWidth, width)
t.set(Tag.ImageLength, height)
t.set(Tag.TileWidth, width)
t.set(Tag.TileLength, height)
t.set(Tag.Orientation, 1)
t.set(Tag.PhotometricInterpretation, 32803)
t.set(Tag.SamplesPerPixel, 1)
t.set(Tag.BitsPerSample, bpp)
t.set(Tag.CFARepeatPatternDim, [2,2])
t.set(Tag.CFAPattern, [0, 1, 1, 2])
t.set(Tag.BlackLevel, 63)
t.set(Tag.WhiteLevel, 1023)
t.set(Tag.ColorMatrix1, ccm1)
t.set(Tag.CalibrationIlluminant1, 21)
t.set(Tag.AsShotNeutral, ([1,1],[1,1],[1,1]))
t.set(Tag.DNGVersion, [1, 4, 0, 0])
t.set(Tag.DNGBackwardVersion, [1, 0, 2, 0])
t.set(Tag.Make, "Camera Brand")
t.set(Tag.Model, "Camera Model")
t.set(Tag.PreviewColorSpace, 2)

RAW2DNG().convert(rawImage, tags=t, filename="3840", path="")
```
[image]

dustinkerstein, 2021-10-07:
Just did some quick testing / comparison between utility.py's conversion of the sample imx477.jpg. It looks like the code above takes ~3x as much time to create the DNG compared to utility.py. Do you happen to know why that might be? Thanks again for your work on this. Will try to get a sensor full resolution output over the next day or so.

dustinkerstein, 2021-10-12:
FYI, here's an ugly test 4032x3040 raw capture that converts to DNG via the above code. Code looks like it's working well. This was accomplished using a custom driver from Arudcam that got the 4032x3040 resolution working at 15fps (it would hang at 30fps). Would it be okay with you to cross post this over to the Nvidia forums?

[out.raw.zip](https://github.com/schoolpost/PiDNG/files/7329764/out.raw.zip)

closed 2022-05-02 by schoolpost