schoolpost/PiDNG #35: Make installation with pip easier

anuejn, 2021-01-15:
Hey, it would be nice if there was an easyer way to install this via pip and other depenpedcy management solutions like pipenv.
Currently one has to install this as `'git+https://github.com/schoolpost/PyDNG.git#subdirectory=src&egg=pydng'`. However it would be nice if the setup.py file would be moved up one level so that one could omit the subdirectory and egg parameters altogether.

Also publishing a PyPy package would be nice?

What are your thoughts?

joaolisboa, 2021-05-30:
Adding my vote to add this as well. I'm currently developing an app for the rpi camera and it would be helpful if this was available as a package to add as a dependency.

schoolpost, 2021-05-31:
> Adding my vote to add this as well. I'm currently developing an app for the rpi camera and it would be helpful if this was available as a package to add as a dependency.

if you can set that up and create a PR I can review/merge it.

volkerjaenisch, 2021-07-15:
@anuejn 

> Hey, it would be nice if there was an easyer way to install this via pip and other depenpedcy management solutions like pipenv.

I fully agree. Made pull request.

Also the package can now be deployed with Poetry (or setuptools).

> Also publishing a PyPy package would be nice?

That would be a problem since an other Package named "pydng" is already on PyPI.  But this pydng is some dogger util.

Therefore I would propose to rename the package to 

* PiDNG
* RPiDNG

or the like.

Cheers,
Volker

schoolpost, 2021-07-16:
Thanks for your work @volkerjaenisch in modifying the library and making the necessary changes to have this library easier to install. Very much appreciated!

closed 2021-07-16 by schoolpost

volkerjaenisch, 2021-07-16:
Thank you @schoolpost for this fine library.