schoolpost/PiDNG #49: Save as Tiff instead of DNG

negativesupply, 2021-08-22:
Your program is working very well, however I have issues using DNG in Open CV with the 32 bit Pi OS (though this is solved with LibRaw on 64 bit Pi OS). 

Would it be possible to have an option to save as Tiff instead of DNG? I am converting raw data from IMX477.

schoolpost, 2021-08-23:
Do you mean you just need a different file extension? DNG is fundamentally a TIFF file already, just has metadata to mark itself as a RAW/DNG file. 

Or do you mean you want de-bayered raw output?

negativesupply, 2021-08-24:
Yes that's correct. For some reason I can't get Open CV to work with DNG, however I may just be approaching it wrong. This plugin, however has been great, but being able output Tiff would be a nice addition. Unless maybe I'm missing that I can simply change the file extension?

volkerjaenisch, 2021-08-24:
@negativesupply 
Just to give a bit more context. The aim of PiDNG is to deliver a mostly consistent raw image for further RAW development.

This is done by a little preprocessing of the RPi raw image and then a lot of crucial enrichment with:

- Black point
- White balance
- Two color correction matrixes (one for 3000 and one for 5800 Kelvin)

If you run the DNG file from PiDNG through e.g. dcraw (utilizing python rawpy) you will get a high quality foto in any format you desire, even TIFF.

But, PiDNG covers a large range of color temperature 3000-5800 Kelvin with only two matrixes. 

If you need more accuracy for your colours you will have to use a colour checker card for calibration.

You may also have a look at  
[](https://www.strollswithmydog.com/open-raspberry-pi-high-quality-camera-raw/)
There is the real magic documented.

Cheers,
Volker

negativesupply, 2021-08-24:
Thank you for providing this context. I believe your program is performing these tasks incredibly well. The DNG files are fantastic, however I'm currently limited to 32 bit Pi OS because of the lack of Pi camera support on the beta 64 but build, and can't use rawpy on 32 bit because of the lack of support. Thus why I'm inquiring about a version of this program that saves to Tiff (which open cv can work with on 32 or 64 bit). Thank you again for your insight.

closed 2021-09-13 by schoolpost