schoolpost/PiDNG #50: RPICAM2DNG() not converting mode 2 High Quality Cam

dgcarr, 2021-09-11:
Get the following error when trying to extract raw DNG from photos taken in mode 2 on rpi high quality camera 
(raspistill --mode 2 -r -o test.jpg):
[image]

Traceback (most recent call last):
    d.convert(im)
  File "C:\Users\daggy\AppData\Roaming\Python\Python39\site-packages\pidng\core.py", line 267, in convert
    rawFrame = self.__process__(image, process)
  File "C:\Users\daggy\AppData\Roaming\Python\Python39\site-packages\pidng\core.py", line 232, in __process__
    rawImage = self.__extractRAW__(input_file)
  File "C:\Users\daggy\AppData\Roaming\Python\Python39\site-packages\pidng\core.py", line 195, in __extractRAW__
    assert data[:4] == 'BRCM'.encode("ascii")
AssertionError

schoolpost, 2021-09-11:
Can you send a link to this RAW image?

dgcarr, 2021-09-13:
Thanks for spending time looking into this, the example image link is here:https://github.com/dgcarr/rawimage/blob/main/image-0000354.jpg  best regardsdgcarr Sent from Mail for Windows From: Csaba NagySent: 12 September 2021 06:58To: schoolpost/PiDNGCc: David Carr; AuthorSubject: Re: [schoolpost/PiDNG] RPICAM2DNG() not converting mode 2 High Quality Cam (#50) Can you send a link to this RAW image?

schoolpost, 2021-09-13:
The reason for this error because even though you are using mode 2 RAW, the jpeg file is still creating a 4056x3040 sized image and therefor PiDNG assumes it is full size RAW. 

So two ways around this:
1. Make the JPEG output the same dimensions as the RAW output ( so 2028x1520 ) 
2. Edit the EXIF data of the jpeg and change the ImageWidth and ImageLength to 2028x1520
3. Edit the PiDNG source and set the version variable to version 4.
https://github.com/schoolpost/PiDNG/blob/30bee229654f07532a473963e7c704a4a8e1af04/pidng/core.py#L178-L183

So you could add a line and just set "ver = 4"

dgcarr, 2021-09-13:
Awesome, thanks! It's a good workaround

closed 2021-09-13 by schoolpost