schoolpost/PiDNG #52: Illegal Instruction

eat-sleep-code, 2022-01-04:
@schoolpost, well this is a weird one.

I am running into an issue with PiDNG, but only on one Rasperry Pi.

This is a Raspberry Pi Zero W with an __Arducam__ IMX477 HQ Camera module.

I ran `pip3 install PiDNG` and it says it successfully installed:  

> Successfully installed PiDNG-3.4.7 exifread-2.3.2 numpy-1.21.5

But then my Python script fails on this line with an error of _Illegal instruction_:

`from pidng.core import RPICAM2DNG`

This worked fine on another Raspberry Pi 4 with a Raspberry Pi-branded HQ camera.    So, not sure if this is related to the Pi Zero W or the Arducam camera.    There is nothing dumped to the screen other than _Illegal instruction_ so not much helpful debugging info.

schoolpost, 2022-01-04:
Can you post the image you are trying to convert?

eat-sleep-code, 2022-01-04:
I never even get that far.   It fails on the call to import the library.   :-)

This is for this repo: https://github.com/eat-sleep-code/camera.remote

It never has a chance to get to the method where it is called: https://github.com/eat-sleep-code/camera.remote/blob/0eb5174d46d9b49ab3cf04c479f27765c75c40aa/camera.py#L279

... because the import statement fails here: https://github.com/eat-sleep-code/camera.remote/blob/0eb5174d46d9b49ab3cf04c479f27765c75c40aa/camera.py#L3

eat-sleep-code, 2022-01-05:
@schoolpost  I have tried uninstalling, deleting the __pycache__, reinstalling,  and still the same issue.   It fails on the import.

eat-sleep-code, 2022-01-05:
@schoolpost I did some more digging here.   

I have successfully deploy the software on two different Raspberry Pi 4's with Raspberry Pi foundation HQ cameras.   Both work without issue.

However, I have tried with three different Raspberry Pi Zero W (original W's, not the new Zero 2) and all three have the same error occurring.   Details of the steps I tried on all three...

---

- From macOS, open Raspberry Pi Imager v1.6.2
    - Click _Choose OS_, select Raspberry Pi OS (other), and then select Raspberry Pi OS Lite (Legacy)
    - Click _Choose Storage_, and select a new 64GB SanDisk Extreme card
    - Press _Cmd-Shift-X_, to show the hidden Raspberry Pi Image menu
        - Set a hostname of `camera-zero.local`
        - Check the _Enable SSH_ box and set a password under the _Password authentication_ section
        - Check the _Configure WiFi_ box and then enter a SSID and password.   Select "US" for the _WiFi country_
        - Check _Set locale settings_ and select "America/Phoenix" for the time zone and "us" for the keyboard layout
        - Click the _Save_ button
    - Click the _Write_ button and confirm
    - When write is complete and verified remove the micro SD card from reader
- Insert the card into the Pi Zero W
- Provide power to the Pi Zero W and wait for it to boot
- From the computer `sudo ssh pi@192.168.1.19` and login with the credentials provided in the _Enable SSH_ section above
- Run `sudo raspi-config`, select _Interface options_, then _Camera_, then _Yes_, and then finally _OK_ to enable the Camera
- Select _Finish_ and then confirm _Yes_ to reboot
- Reconnect with SSH
- Execute `sudo apt update && sudo apt full-upgrade -y && sudo apt autoremove -y && sudo apt autoclean` to do a full update of the default installed packages
    - The __rpi-eeprom__ is the only package that is updated
- Execute the following commands:
```bash
    wget -q https://raw.githubusercontent.com/eat-sleep-code/camera.remote/main/install-camera.sh -O ~/install-camera.sh
    sudo chmod +x ~/install-camera.sh && ~/install-camera.sh
```

- Below is the console output
    - Unfortunately, executing camera.remote still raises an 'Illegal instruction' on this line `from pidng.core import RPICAM2DNG` 

```bash
    
Camera Remote [Installation Script] 
-------------------------------------------------------------------------- 

Updating package repositories... 
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists...
Building dependency tree...
Reading state information...
All packages are up to date.

Installing prerequisites... 
Reading package lists...
Building dependency tree...
Reading state information...
python3 is already the newest version (3.7.3-1).
python3 set to manually installed.
The following additional packages will be installed:
  dh-python gir1.2-glib-2.0 git-man libatlas3-base libcurl3-gnutls
  liberror-perl libexpat1-dev libgfortran5 libgirepository-1.0-1
  libpython3-dev libpython3.7-dev python-pip-whl python3-asn1crypto
  python3-cffi-backend python3-crypto python3-cryptography python3-dbus
  python3-dev python3-distutils python3-entrypoints python3-gi python3-keyring
  python3-keyrings.alt python3-lib2to3 python3-numpy python3-secretstorage
  python3-setuptools python3-wheel python3-xdg python3.7-dev
Suggested packages:
  git-daemon-run | git-daemon-sysvinit git-doc git-el git-email git-gui gitk
  gitweb git-cvs git-mediawiki git-svn libatlas-doc liblapack-doc
  python-crypto-doc python-cryptography-doc python3-cryptography-vectors
  python-dbus-doc python3-dbus-dbg gnome-keyring libkf5wallet-bin
  gir1.2-gnomekeyring-1.0 gfortran python-numpy-doc python3-pytest
  python3-numpy-dbg python-picamera-docs python-secretstorage-doc
[... 214 lines cut ...]
Collecting sysv-ipc>=1.1.0 (from Adafruit-Blinka->adafruit-circuitpython-neopixel)
  Downloading https://www.piwheels.org/simple/sysv-ipc/sysv_ipc-1.1.0-cp37-cp37m-linux_armv6l.whl (68kB)
Collecting pyserial>=3.0 (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-neopixel)
  Downloading https://files.pythonhosted.org/packages/07/bc/587a445451b253b285629263eb51c2d8e9bcea4fc97826266d186f96f558/pyserial-3.5-py2.py3-none-any.whl (90kB)
Collecting pyusb!=1.2.0,>=1.0.0 (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-neopixel)
  Downloading https://files.pythonhosted.org/packages/15/a8/4982498b2ab44d1fcd5c49f07ea3795eab01601dc143b009d333fcace3b9/pyusb-1.2.1-py3-none-any.whl (58kB)
Installing collected packages: RPi.GPIO, Adafruit-PureIO, pyserial, pyusb, pyftdi, rpi-ws281x, Adafruit-PlatformDetect, sysv-ipc, Adafruit-Blinka, adafruit-circuitpython-pixelbuf, adafruit-circuitpython-neopixel, numpy, exifread, PiDNG
  Found existing installation: RPi.GPIO 0.7.0
    Not uninstalling rpi.gpio at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'RPi.GPIO'. No files were found to uninstall.
  Found existing installation: numpy 1.16.2
    Not uninstalling numpy at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'numpy'. No files were found to uninstall.
Successfully installed Adafruit-Blinka-6.17.0 Adafruit-PlatformDetect-3.19.1 Adafruit-PureIO-1.1.9 PiDNG-3.4.7 RPi.GPIO-0.7.0 adafruit-circuitpython-neopixel-6.2.3 adafruit-circuitpython-pixelbuf-1.1.1 exifread-2.3.2 numpy-1.21.5 pyftdi-0.53.3 pyserial-3.5 pyusb-1.2.1 rpi-ws281x-4.3.1 sysv-ipc-1.1.0

Installing Camera Remote... 

Downloading color profiles... 

Setting up alias... 
Please ensure that your camera and I2C interfaces are enabled in raspi-config before proceeding.

-------------------------------------------------------------------------- 
Installation completed. 
```

eat-sleep-code, 2022-01-06:
The best thing I can guess at this point is that you have something that is not supported by the BCM2835 ARMv6 chips in the Zero W (vs the BCM2711 ARMv8 SoC in the Pi 4).   Any chance we can get a fix?

eat-sleep-code, 2022-01-07:
OMG, this may have nothing to do with PiDNG.   It appears a recent update to the "Legacy" Buster has broken a bunch of stuff on Pi Zero's.    Folks getting this error with Chrome, etc.

eat-sleep-code, 2022-01-07:
@schoolpost,  I have tracked the root cause of this issue...

NumPy 1.21.5 was released on December 20th.  It appears to have introduced breaking changes on 32-bit ARMV6 processors.

As your install script has the dependency restriction to be any version under 2.0 (`Collecting numpy<2.0,>=1.14 (from PiDNG)`) it grabs 1.21.5 during the install.   This works find on the 64-bit ARMV8 processors, but does not work on the "older" Raspberry Pi Zero W.

It appears NumPy 1.21.5 was a maintenance release, so not sure why it ended up causing this breaking change.   It also appears 1.22.0 dropped support for 32-bit processors so all future versions will likely not work on the original Pi Zero W.   

To "resolve" this, I ran the following command after installing PiDNG.   This reverts NumPy to use the older -- working -- 1.21.4 version.

`sudo pip3 uninstall -y numpy && sudo pip3 install numpy==1.21.4`

eat-sleep-code, 2022-01-07:
I have opened an [issue](https://github.com/numpy/numpy/issues/20765) with NumPy on this to see if they will resolve, or if support for the original Pi Zero W is considered "dead" by them.

eat-sleep-code, 2022-01-13:
Just as an update...

I worked quite a bit with the developers of NumPy to find what the root cause of it was.  

After much digging, this was traced to an issue with PiWheels affecting all Python 3.9 wheels that were touched in the last several weeks.   

The creator of PiWheels posted that this is currently being researched: https://github.com/piwheels/packages/issues/269#issuecomment-1007273390

As of the morning of January 13, 2022 there is no resolution to this.

closed 2022-05-02 by schoolpost