schoolpost/PiDNG #54: struct.error: unpack requires a buffer of 4380480 bytes

kyleboddy, 2022-01-17:
Trying this code and getting the following output (width/height are correct):

```
from pidng.core import RAW2DNG, DNGTags, Tag
import numpy as np
import struct, collections

width = 2028
height = 1080
bpp=10
numPixels = width*height
rawFile = 'picture.raw'
rf = open(rawFile, mode='rb')
rawData = struct.unpack("H"*numPixels,rf.read(2*numPixels))
rawFlatImage = np.zeros(numPixels, dtype=np.uint16)
rawFlatImage[:] = rawData[:]
rawImage = np.reshape(rawFlatImage,(height,width))

ccm1 = [[19549, 10000], [-7877, 10000], [-2582, 10000],
        [-5724, 10000], [10121, 10000], [1917, 10000],
        [-1267, 10000], [ -110, 10000], [ 6621, 10000]]

t = DNGTags()
t.set(Tag.ImageWidth, width)
t.set(Tag.ImageLength, height)
t.set(Tag.TileWidth, width)
t.set(Tag.TileLength, height)
t.set(Tag.Orientation, 1)
t.set(Tag.PhotometricInterpretation, 32803)
t.set(Tag.SamplesPerPixel, 1)
t.set(Tag.BitsPerSample, bpp)
t.set(Tag.CFARepeatPatternDim, [2,2])
t.set(Tag.CFAPattern, [0, 1, 1, 2])
t.set(Tag.BlackLevel, 63)
t.set(Tag.WhiteLevel, 1023)
t.set(Tag.ColorMatrix1, ccm1)
t.set(Tag.CalibrationIlluminant1, 21)
t.set(Tag.AsShotNeutral, ([1,1],[1,1],[1,1]))
t.set(Tag.DNGVersion, [1, 4, 0, 0])
t.set(Tag.DNGBackwardVersion, [1, 0, 2, 0])
t.set(Tag.Make, "Camera Brand")
t.set(Tag.Model, "Camera Model")
t.set(Tag.PreviewColorSpace, 2)

RAW2DNG().convert(rawImage, tags=t, filename="pythonoutput", path="")
```

Output of `python dngconvert.py`:

```
Traceback (most recent call last):
  File "/home/pi/dngconvert.py", line 11, in <module>
    rawData = struct.unpack("H"*numPixels,rf.read(2*numPixels))
struct.error: unpack requires a buffer of 4380480 bytes
```

Not sure how to proceed from here - did a bit of Googling without a lot of help there. Thanks!

kyleboddy, 2022-01-17:
Sample post-processed photo via dcraw:

[image]

RAW file that produced this (as well as the errors above) attached below:

[picture raw.zip](https://github.com/schoolpost/PiDNG/files/7879264/picture.raw.zip)

closed 2022-05-02 by schoolpost

willswordh, 2023-10-17:
@kyleboddy Hi, how did you manage to solve the issue?