schoolpost/PiDNG #66: Can't install PiDNG

Luffitys, 2022-07-19:
LOG:

```
C:\Users\Daniel\Downloads\PiDNG-master>python -mpip install PiDNG
Collecting PiDNG
  Using cached pidng-4.0.9.tar.gz (21 kB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: numpy in c:\users\daniel\appdata\local\programs\python\python38\lib\site-packages (from PiDNG) (1.20.3)
Building wheels for collected packages: PiDNG
  Building wheel for PiDNG (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [73 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build\lib.win-amd64-cpython-38
      creating build\lib.win-amd64-cpython-38\pidng
      copying src\pidng\camdefs.py -> build\lib.win-amd64-cpython-38\pidng
      copying src\pidng\core.py -> build\lib.win-amd64-cpython-38\pidng
      copying src\pidng\defs.py -> build\lib.win-amd64-cpython-38\pidng
      copying src\pidng\dng.py -> build\lib.win-amd64-cpython-38\pidng
      copying src\pidng\legacy.py -> build\lib.win-amd64-cpython-38\pidng
      copying src\pidng\packing.py -> build\lib.win-amd64-cpython-38\pidng
      copying src\pidng\__init__.py -> build\lib.win-amd64-cpython-38\pidng
      running egg_info
      writing src\pidng.egg-info\PKG-INFO
      writing dependency_links to src\pidng.egg-info\dependency_links.txt
      writing requirements to src\pidng.egg-info\requires.txt
      writing top-level names to src\pidng.egg-info\top_level.txt
      reading manifest file 'src\pidng.egg-info\SOURCES.txt'
      reading manifest template 'MANIFEST.in'
      adding license file 'LICENSE'
      writing manifest file 'src\pidng.egg-info\SOURCES.txt'
      C:\Users\Daniel\AppData\Local\Programs\Python\Python38\lib\site-packages\setuptools\command\build_py.py:153: SetuptoolsDeprecationWarning:     Installing 'pidng.liblj92' as data is deprecated, please list it in `packages`.
          !!

          ############################
          # Package would be ignored #
          ############################
          Python recognizes 'pidng.liblj92' as an importable package,
          but it is not listed in the `packages` configuration of setuptools.

          'pidng.liblj92' has been automatically added to the distribution only
          because it may contain data files, but this behavior is likely to change
          in future versions of setuptools (and therefore is considered deprecated).

          Please make sure that 'pidng.liblj92' is included as a package by using
          the `packages` configuration field or the proper discovery methods
          (for example by using `find_namespace_packages(...)`/`find_namespace:`
          instead of `find_packages(...)`/`find:`).

          You can read more about "package discovery" and "data files" on setuptools
          documentation page.

      !!

        check.warn(importable)
      copying src\pidng\bitunpack.c -> build\lib.win-amd64-cpython-38\pidng
      creating build\lib.win-amd64-cpython-38\pidng\liblj92
      copying src\pidng\liblj92\lj92.c -> build\lib.win-amd64-cpython-38\pidng\liblj92
      copying src\pidng\liblj92\lj92.h -> build\lib.win-amd64-cpython-38\pidng\liblj92
      running build_ext
      building 'ljpegCompress' extension
      creating build\temp.win-amd64-cpython-38
      creating build\temp.win-amd64-cpython-38\Release
      creating build\temp.win-amd64-cpython-38\Release\src
[... 88 lines cut ...]
      "C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.32.31326\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -IC:\Users\Daniel\AppData\Local\Programs\Python\Python38\include -IC:\Users\Daniel\AppData\Local\Programs\Python\Python38\Include "-IC:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.32.31326\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\shared" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\um" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\winrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\cppwinrt" /Tcsrc/pidng/bitunpack.c /Fobuild\temp.win-amd64-cpython-38\Release\src/pidng/bitunpack.obj -std=gnu99
      cl : Command line warning D9002 : ignoring unknown option '-std=gnu99'
      bitunpack.c
      "C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.32.31326\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -IC:\Users\Daniel\AppData\Local\Programs\Python\Python38\include -IC:\Users\Daniel\AppData\Local\Programs\Python\Python38\Include "-IC:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.32.31326\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\shared" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\um" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\winrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.19041.0\\cppwinrt" /Tcsrc/pidng/liblj92/lj92.c /Fobuild\temp.win-amd64-cpython-38\Release\src/pidng/liblj92/lj92.obj -std=gnu99
      cl : Command line warning D9002 : ignoring unknown option '-std=gnu99'
      lj92.c
      src/pidng/liblj92/lj92.c(226): warning C4334: '<<': result of 32-bit shift implicitly converted to 64 bits (was 64-bit shift intended?)
      src/pidng/liblj92/lj92.c(761): warning C4013: '__builtin_clz' undefined; assuming extern returning int
      src/pidng/liblj92/lj92.c(794): warning C4244: 'initializing': conversion from 'int' to 'float', possible loss of data
      "C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.32.31326\bin\HostX86\x64\link.exe" /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:EMBED,ID=2 /MANIFESTUAC:NO /LIBPATH:C:\Users\Daniel\AppData\Local\Programs\Python\Python38\libs /LIBPATH:C:\Users\Daniel\AppData\Local\Programs\Python\Python38 /LIBPATH:C:\Users\Daniel\AppData\Local\Programs\Python\Python38\PCbuild\amd64 "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.32.31326\lib\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\lib\um\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.19041.0\ucrt\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\\lib\10.0.19041.0\\um\x64" /EXPORT:PyInit_ljpegCompress build\temp.win-amd64-cpython-38\Release\src/pidng/bitunpack.obj build\temp.win-amd64-cpython-38\Release\src/pidng/liblj92/lj92.obj /OUT:build\lib.win-amd64-cpython-38\ljpegCompress.cp38-win_amd64.pyd /IMPLIB:build\temp.win-amd64-cpython-38\Release\src/pidng\ljpegCompress.cp38-win_amd64.lib
         Creating library build\temp.win-amd64-cpython-38\Release\src/pidng\ljpegCompress.cp38-win_amd64.lib and object build\temp.win-amd64-cpython-38\Release\src/pidng\ljpegCompress.cp38-win_amd64.exp
      lj92.obj : error LNK2001: unresolved external symbol __builtin_clz
      build\lib.win-amd64-cpython-38\ljpegCompress.cp38-win_amd64.pyd : fatal error LNK1120: 1 unresolved externals
      error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2022\\BuildTools\\VC\\Tools\\MSVC\\14.32.31326\\bin\\HostX86\\x64\\link.exe' failed with exit code 1120
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> PiDNG

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
```

wgprojects, 2022-07-20:
I am having the same problem!
lj92.obj : error LNK2001: unresolved external symbol __builtin_clz

wgprojects, 2022-07-20:
Workaround: I modified lj92.c in my own fork to remove the builtin. Install with:
python -mpip install  git+https://github.com/wgprojects/PiDNG.git

Luffitys, 2022-07-21:
@wgprojects If no issues are caused by your commits, a pull request would probably be nice

wgprojects, 2022-07-21:
I'm not super familiar with PiDNG, I don't feel I'm able to prove it's a good enough fix for a PR.
I found a few lj92.c online which were substantially different; it feels like the change was intentional for performance reasons. I'm sure there's a better way to get gcc to use the builtin when it's available and fall back to the looping version...

unpapardo, 2022-07-23:
> Workaround: I modified lj92.c in my own fork to remove the builtin. Install with: python -mpip install git+https://github.com/wgprojects/PiDNG.git

Can confirm it worked in my Win10

teerytko, 2022-10-17:
I did not notice your workaround and created my own, which is also a pull request https://github.com/schoolpost/PiDNG/pull/67

wgprojects, 2022-10-17:
Thanks for cleaning my workaround into an acceptable commit!

DIYer22, 2023-01-11:
Same error in a clean docker image based on Ubuntu20.04 .
Fix the issue by temprory add Python.h by install python3-dev
```dockerfile
RUN apt-get install -y python3-dev && \
    pip install pidng && \
    apt-get remove -y python3-dev && \
    apt-get autoremove -y && \
    apt-get clean
```

luketaverne, 2023-02-28:
The current 4.0.9 release on pypi is still broken on Windows. Installing directly from this repo with `pip install git+https://github.com/schoolpost/PiDNG.git` works. Would you be able to make a new release on pypi? Thanks in advance!

p0las, 2024-01-24:
unfortunately sill broken. it works from the alternate repo. would be nice to merge them
`pip install git+https://github.com/schoolpost/PiDNG.git`

ai2ys, 2024-03-06:
I was not able to install the latest version of pidng and got the same error as mentioned in this issue using both installation methods below when trying to install `pidng` in a Docker image based on `python:3.11-slim-bookworm`.

- `pip install pidng==4.0.9`
- `pip install git+https://github.com/schoolpost/PiDNG.git` 

But both installation methods work when using `python:3.11-bookworm` (not the `slim` version) instead.

hkoren, 2024-11-13:
this fixed PiDNG install for me, recommend making it PR to the project

wgprojects, 2024-11-13:
Feel free!

Yahoo Mail: Search, Organize, Conquer