schoolpost/PiDNG #70: raw2dng produce a DNG image which don't open in Adobe Lightroom

yoav-chai, 2022-12-16:
Hey,
I used the raw2jpg.py (from the examples directory), which runs the raw file located in the extra directory, and the DNG file that its process does not open up in Adobe Lightroom (or any other Adobe product).
Is there a way to make the DNG file valid so I can open it up in Adobe products?

Thanks!

schoolpost, 2022-12-25:
Hello,

Please run the dng file through adobe-dng-validate tool ( can be found online ) to determine why the dng file is not working.

yoav-chai, 2022-12-26:
Hey,
I got the below error, so I removed the tiles tag, and now the dng validate is working.
Thanks

Validating "custom.dng"...
*** Warning: IFD 0 tags are not sorted in ascending numerical order ***
*** Warning: IFD 0 tags are not sorted in ascending numerical order ***
*** Warning: Missing or invalid UniqueCameraModel ***
*** Error: IFD uses both strips and tiles (IFD 0) ***

closed 2022-12-26 by yoav-chai