schoolpost/PiDNG #71: monochrome pixel format support

fiepfiep, 2023-01-16:
I believe currently it is not possible to convert raw files to monochrome DNG files. (it is just a tag I assume, but will require some change in the code)
what exactly would need to change here? is anyone working on it? If needed I can provide some image files captured with a monochrome image sensor.

schoolpost, 2023-01-20:
I have not tested / experimented with monochrome output. You will need to change this tag from it's current value which is set to: CFA ( https://www.awaresystems.be/imaging/tiff/tifftags/photometricinterpretation.html ) 

There may be some others, but I can't say for certain without testing.

sschmaus, 2024-09-18:
Commenting in case anyone else runs into the same issue, for monochrome DNGs, the following tags need to be set: 
```python
t.set(Tag.PhotometricInterpretation, PhotometricInterpretation.Linear_Raw)
t.set(Tag.SamplesPerPixel, 1)
t.set(Tag.BitsPerSample, 16) #the bit depth of your image may differ
```

Tags like `ColorMatrix1` or `AsShotNeutral` are not needed with monochrome data.

closed 2026-11-06 by phili-b