schoolpost/PiDNG #75: Produces black(ish) output dng file

kevflynn, 2023-05-18:
I am trying to convert a raw photo using the following code (from the example) 
`from pidng.core import RAW2DNG, DNGTags, Tag
from pidng.defs import *
import numpy as np
import struct
import rawpy
import sys

# load raw data into 16-bit numpy array.
raw_file = sys.argv[1]
output_file = sys.argv[2]

with rawpy.imread(raw_file) as raw:
    width, height = raw.raw_image.shape
    bit_depth = raw.raw_image.dtype.itemsize * 8  # bytes to bits
    raw_image = raw.raw_image

num_pixels = width * height

# uncalibrated color matrix, just for demo.
ccm1 = [
    [19549, 10000],
    [-7877, 10000],
    [-2582, 10000],
    [-5724, 10000],
    [10121, 10000],
    [1917, 10000],
    [-1267, 10000],
    [-110, 10000],
    [6621, 10000],
]

# set DNG tags.
t = DNGTags()
t.set(Tag.ImageWidth, width)
t.set(Tag.ImageLength, height)
t.set(Tag.TileWidth, width)
t.set(Tag.TileLength, height)
t.set(Tag.Orientation, Orientation.Horizontal)
t.set(Tag.PhotometricInterpretation, PhotometricInterpretation.Color_Filter_Array)
t.set(Tag.SamplesPerPixel, 1)
t.set(Tag.BitsPerSample, bit_depth)
t.set(Tag.CFARepeatPatternDim, [2, 2])
t.set(Tag.CFAPattern, CFAPattern.GBRG)
t.set(Tag.BlackLevel, (4096 >> (16 - bit_depth)))
t.set(Tag.WhiteLevel, ((1 << bit_depth) - 1))
t.set(Tag.ColorMatrix1, ccm1)
t.set(Tag.CalibrationIlluminant1, CalibrationIlluminant.D65)
t.set(Tag.AsShotNeutral, [[1, 1], [1, 1], [1, 1]])
t.set(Tag.BaselineExposure, [[-150, 100]])
t.set(Tag.Make, "Camera Brand")
t.set(Tag.Model, "Camera Model")
t.set(Tag.DNGVersion, DNGVersion.V1_4)
t.set(Tag.DNGBackwardVersion, DNGVersion.V1_2)
t.set(Tag.PreviewColorSpace, PreviewColorSpace.sRGB)

# save to dng file.
r = RAW2DNG()
r.options(t, path="", compress=True)
r.convert(raw_image, filename=output_file)
`

and it's producing an image - but the image is just a black blur. I suspect it's because I'm not changing the color matrix settings? I don't know how to configure that - any guidance on that? or anything else that could be causing this?

schoolpost, 2023-05-21:
Can you include a screenshot or sample of the image? What camera is it from?

IDKiro, 2023-07-26:
Same problem here, I ran the provided example directly, raw2dng.py, and it successfully reads the image and stores it as custom.dng, but using various tools it reads an all-black image (including using rawpy).

OParczyk, 2024-03-04:
I can confirm this issue on debian12 too, both with python3.11 and 3.12.
used PiDNG from GitHub too.
numpy==1.26.4

[raw2dng.py](https://github.com/schoolpost/PiDNG/blob/master/examples/raw2dng.py) with the supplied [scene_daylight_211ms_c2.raw16](https://github.com/schoolpost/PiDNG/blob/master/extras/scene_daylight_211ms_c2.raw16) yields a completely black image.

OParczyk, 2024-03-04:
So I just came around to fiddle with it again and I have an interesting observation:
Changing `r.options(t, path="", compress=True)` to `r.options(t, path="", compress=False)`
Makes it work again! (but weirdly to me the resulting .dng file is still only 50% of the original raw filesize, hmm)

For now that'll do as a workaround for me, I'll report back should I find something else regarding this.

sgarg367, 2025-04-11:
@OParczyk Tried your change, but now it generates a corrupted output.
Output file size is increased to 18.5 MB, but still can't open it.

Any luck in making this code work ?