schoolpost/PiDNG #79: Support for GPS tags

MadsDyrmann, 2023-10-14:
I really appreciate this library, and it functions effectively for exporting uncompressed DNGs with various tags. However, I have a camera equipped with a GPS module, and I would greatly appreciate it if it were also possible to add GPS tags to the DNG files. I am uncertain whether this is already achievable through a sub-IFD. If it is, could you kindly provide guidance on how to so so?

fwachsmuth, 2023-10-14:
I'd recommend a look into exiftool for that.

MadsDyrmann, 2023-10-14:
I am actually using exiftool for that, which works. But it seems like an extra step first exporting the dng file using PiDNG, then open and exporting it again using exiftool. It would be ideal, if the GPS tags could be added together with the other tags that PiDNG adds in the first place.

MadsDyrmann, 2023-10-14:
I just did a test: Exporting a 12Mp image (8 bit bayer-encoded numpy array) to DNG takes ~0,015s when using PiDNG. Adding GPS data using exiftool takes additionally ~0.188s.
It would therefore reduce the writing time significantly, if GPS data could be added from PiDNG.