schoolpost/PiDNG #85: how to save valid dng from nparray?

vladmandic, 2024-09-15:
if i already have correct hdr image in `np.array`, how do i save it to **dng** file?
assuming `shape=h, w, 3; dtype=uint16`

i've tried all different combos i can think of and tested using:
- adobe's `dng_validate`
- adobe photoshop
- windows photos

tested:
```log
- BitsPerSample=16          PhotometricInterpretation=Linear_Raw        SamplesPerPixel=1  
  vaidation pass, ps unprocessed monochrome, windows fails  
- BitsPerSample=16          PhotometricInterpretation=Linear_Raw        SamplesPerPixel=3  
  validation fail invalid SamplesPerPixel, ps fails, windows overexposes  
- BitsPerSample=16,16,16    PhotometricInterpretation=Linear_Raw        SamplesPerPixel=3  
  validation fail invalid SamplesPerPixel, ps fails, windows overexposes  
- BitsPerSample=16,16,16    PhotometricInterpretation=Linear_Raw        SamplesPerPixel=1  
  validation fail invalid SamplesPerPixel, ps fails windows fails  
- BitsPerSample=16,16,16    PhotometricInterpretation=RGB               SamplesPerPixel=1  
  validation fail PhotometricInterpretation requires NewSubFileType, ps fails, windows fails  
- BitsPerSample=16          PhotometricInterpretation=RGB               SamplesPerPixel=3  
  validation fail PhotometricInterpretation requires NewSubFileType, ps fails, windows pass  
```

as far as i understand, PhotometricInterpretation.RGB should only be used for previews which are in second block, thus the error.
but i cannot figure out Linear_Raw format?

thanks for the assistance!

sschmaus, 2024-09-18:
Assuming you are working with already debayered raw sensor data strored in an ndarray with `shape=h, w, 3; dtype=uint16`, the correct tags would be `BitsPerSample=16,16,16    PhotometricInterpretation=Linear_Raw        SamplesPerPixel=3`

There are a few possibilities why your code might not create dng files that are supported by Adobe software and this is very difficult to debug since Adobe doesn't give you good debug information. 

The first issue might be due to a31edac85985f0fea02c9835b66dd58e4b2c6e32, I believe this change broke compatibility with Adobe software so I forked this repo for my work and reverted the change there: https://github.com/sschmaus/PiDNG/tree/AdobeCompatibility

Another pitfall may be that the [Adobe DNG specification](https://helpx.adobe.com/content/dam/help/en/photoshop/pdf/DNG_Spec_1_7_1_0.pdf) is very strict about which tags are needed if you are working with actual RAW data (usually stored with `PhotometricInterpretation=Linear_Raw` or `PhotometricInterpretation=Color_Filter_Array`)

Since you didn't mention it, make sure you have at least the `ColorMatrix1 `tag defined, otherwise Adobe software will refuse to read your file, while Windows Photo view which is not as strict in enforcing the correct tags may still display an image (though often with very mangled colors)

This tag defines the color transformation from the raw colors that your camera captures to the XYZ color space. It's unique for every camera (model), but if you don't know it four your camera, you could use this one for debugging purposes: https://github.com/schoolpost/PiDNG/blob/66f3e16cf235443e4255a0ad382aa9d27bd1c095/examples/raw2dng.py#L21-L24

vladmandic, 2024-09-18:
using `BitsPerSample=16,16,16    PhotometricInterpretation=Linear_Raw        SamplesPerPixel=3` fail `dng_validate` - and i would 
i expect as such as based on my understanding, it should be either `16 x 3` or `[16,16,16] x 1`
and `ccm` is used/required/needed only if `PhotometricInterpretation=Color_Filter_Array`, its not a valid tag when using `Linear_Raw` - that is whole point of linear_raw.

sschmaus, 2024-09-18:
BitsPerSample should be set to `[16,16,16]` in your case.

As per the [DNG specficication](https://helpx.adobe.com/content/dam/help/en/photoshop/pdf/DNG_Spec_1_7_1_0.pdf) (p. 32): 
> The ColorMatrix1 tag is required for all non-monochrome DNG files.

So `ColorMatrix1` is definitely needed if you are using an RGB image with `PhotometricInterpretation=Linear_Raw`.

You don't need it with `PhotometricInterpretation=RGB`, because that assumes your image is already mapped from the camera`s native color space to sRGB, but as you note this is only intended to be used for previews.

Just to make sure, is your image still in the camera`s native color space or is it already mapped to sRGB or any other color space?

vladmandic, 2024-09-18:
ahhh....seems like i'll have to create a neutral colormatrix then somehow.

image is not from camera at all - its a synthetically generated hdr image, i just want to save it to dng. so yes, its 100% rgb using uint16 range.

sschmaus, 2024-09-19:
>  its a synthetically generated hdr image

oh that certainly changes things a bit. 

Is it already gamma encoded or in a linear brightness range?
You can read [this article](https://blog.johnnovak.net/2016/09/21/what-every-coder-should-know-about-gamma/) to understand what gamma is and why it is important to handle it correctly.
In the following I assume your image is in a linear brightness range. 

The easy option might be be to just import your image as a .tif file in Photoshop, since you probably don't need any of the features of the DNG format. 
Though if you don't specify a color space in the file, Photoshop will assume it is in sRGB color space which requires the brightness to be endoded with the sRGB gamma function. 
[This repository](https://github.com/PetterS/opencv_srgb_gamma) has example functions that you could adapt to work with your 16 bit data. 

If you still want to go the DNG route for some reason, you can set 'ColorMatrix1' to the [XYZ to sRGB matrix](http://www.brucelindbloom.com/index.html?Eqn_RGB_XYZ_Matrix.html) (yes the DNG spec requires the inverse matrix here) to trick Photoshop into interpreting the image as sRGB and not a raw camera image.
Since the DNG spec expects linear color values you don't need to apply gamma here.
```python
#XYZ to RGB matrix to ingest sRGB color-encoded images in the DNG format
ccm1 = [[ 3240454, 1000000], [ -1537138, 1000000], [ -498531, 1000000],
        [ -969266, 1000000], [  1876010, 1000000], [   41556, 1000000],
        [   55643, 1000000], [  -204025, 1000000], [ 1057225, 1000000]]
```

vladmandic, 2024-09-19:
> Is it already gamma encoded or in a linear brightness range?

its in linear brightness range - example repo you've posted helps a lot!

> If you still want to go the DNG route for some reason, you can set 'ColorMatrix1

thanks, that did the trick!

closed 2024-09-19 by vladmandic